fluidd-core/FluiddPI #5: Problem displaying gcode thumbnail generated from Simplifi3D.

doktorcb, 2021-07-26:
I was looking for where the problem with the thumbnail is and why they don't appear, after a long time I discovered this.
There is no option to search for thumbnails in the "metadata.py" configuration file for the S3D slicer. I copied this part from the Prusaslicer and copied it to the Simplifi3D part and everything works as it should. Please integrate this item permanently, there are enough fans who will appreciate this support.

doktorcb, 2021-07-26:
For clarification, this is the file "/home/pi/moonraker/moonraker/components/file_manager/metadata.py"

cadriel, 2021-08-02:
Hi,

This isn't really anything to do with FluiddPI - but sounds like something to do with moonraker. Sugegst posting your issue there. :)

https://github.com/Arksine/moonraker

closed 2021-08-02 by cadriel