fluidd-core/FluiddPI #8: More instructions on how to set up camera?

leandrotoledo, 2021-08-30:
Hi!

I'm using FLuiddPi and trying to set up a camera. I came across this article:
https://docs.fluidd.xyz/features/cameras

But I can't find `/boot/webcam.txt`, or even `/boot/fluiddpi.txt`. Maybe I'm missing something?

Camera type: Logitech C270
Enabled under Settings with Stream typ: MJPEG Adaptive
Camera URL: /webcam/?action=stream

Thanks a lot for this image!

leandrotoledo, 2021-08-31:
I figured out, the webcam.txt is in fact accessible via Fluidd's webinterface. From there I simply uncommented `camera="auto"` and after restarting the host it started working.

closed 2021-08-31 by leandrotoledo

killer1kay, 2023-05-23:
> > To get started, open the webcam.txt config file in Fluidd’s configuration editor. Instructions on how to get started as well as option references are included in this file.
> 
> I might be missing something fundamental here but webcam.txt was not generated in my fresh install of Fluidd [image]

same I don't see that file