fluidd-core/FluiddPI #10: Error when defining CANCEL_PRINT gcode macro in print.cfg: "gcode command CANCEL_PRINT already registered"

boomninjavanish, 2021-09-27:
Hi!

Having only used FluiddPi for a only a few prints, I wasn't sure if this issue belongs here or in one of the other components that make up the project. If someone could point me in the right direction, I would greatly appreciate it!

I have been using FluiddPi with a CANCEL_PRINT gcode macro defined in printer.cfg for a few days. However, my printer.cfg became corrupted after the Raspberry Pi became filled with logs (this was a problem with KlipperScreen). After solving this problem and restoring my printer.cfg from a previous version, I was able to get things running again. 

Today, I updated the all of the components within FluiddPi then set out to rebuild my CANCEL_PRINT macro. However, now it complains that the macro already exists. Commenting out the macro causes the system to complain about the CANCEL_PRINT macro being missing! lol wut?

I tried to comment out the [CANCEL_PRINT] macro present in fluidd.cfg without success as well.

Attached are the log files and my configs. Thanks in advanced and thanks for making such great software!

[klippy.log](https://github.com/cadriel/FluiddPI/files/7238893/klippy.log)
[moonraker.log](https://github.com/cadriel/FluiddPI/files/7238894/moonraker.log)
[fluidd.cfg.txt](https://github.com/cadriel/FluiddPI/files/7238908/fluidd.cfg.txt)
[KlipperScreen.conf.txt](https://github.com/cadriel/FluiddPI/files/7238909/KlipperScreen.conf.txt)
[moonraker.conf.txt](https://github.com/cadriel/FluiddPI/files/7238910/moonraker.conf.txt)
[printer.cfg.txt](https://github.com/cadriel/FluiddPI/files/7238911/printer.cfg.txt)

boomninjavanish, 2021-11-07:
@zenturacp 
Thanks for clarifying. I just fixed based on your hint. I guess I missed the documentation's explanation as I was reading a tutorial that had it setup incorrectly. Thanks again!

zenturacp, 2021-11-08:
> @zenturacp Thanks for clarifying. I just fixed based on your hint. I guess I missed the documentation's explanation as I was reading a tutorial that had it setup incorrectly. Thanks again!

urw! :-) Have a great day