fluidd-core/FluiddPI #18: FluiddPi 'unknown' error, no communication with klippy

bulbersator, 2021-12-07:
Hi, I installed FluidPi on RPI Zero 2W, and I have 'unknown' error. 
On first run I had error about printer.cfg file not present, when I pasted my config file (that was working on the same printer under Octoprint) it changed to 'unknown' error are there seem to be no communication with klippy service. 
It looks like this error is repeating:
`[websockets.py:build_error()] - JSON-RPC Request Error: -32601`
I installed FluidPi once again from ground up and the behavior is the same. 
I also tried to update everything - nothing changes. 

I'm attaching log files: 
[klippy.log.txt](https://github.com/fluidd-core/FluiddPI/files/7671102/klippy.log.txt)
[moonraker.log-1.txt](https://github.com/fluidd-core/FluiddPI/files/7671103/moonraker.log-1.txt)

BeanieBen9990, 2022-02-02:
I'm having this same issue, I have no idea why my configuration wouldn't be working in fluidd but worked fine in octoprint

BeanieBen9990, 2022-02-02:
Ah nevermind, I fixed the problem. I'm not sure if the same steps will work for you, but I made sure to chown everything in the /home/pi directory to pi:pi and then I rebooted. This seemed to fix the problem, but I had the problem again when I turned off the printer and then turned it back on, but was fixed when I ssh'd into the pi and rebooted, so it seems like the problem might arise from either klipper, fluidd, or moonraker booting up before the printer's mainboard does.

closed 2022-03-18 by matmen