fluidd-core/FluiddPI #20: Klipper and Fluiddpi Showing as Insecurity and Malware on Local Network?

Declan0811, 2022-01-05:
I have had fluiddpi installed on my pi for a while now, but now I'm getting alerts for my network that says that Fluiddpi is a network insecurity, and klipper_probe_speed (I just installed a BLTouch) is malware. This isn't really an issue for me, but I'm worried about my ISP, and I want to know how to get these alerts with the pi to stop.

matmen, 2022-02-22:
What network monitoring tool / anti-malware program are you using?
You should get no issues with your ISP, they will never see Fluidd's traffic (unless you run it on a public web server, which I doubt you do)

Declan0811, 2022-02-22:
> What network monitoring tool / anti-malware program are you using?
> 
> You should get no issues with your ISP, they will never see Fluidd's traffic (unless you run it on a public web server, which I doubt you do)

Just a built-in Netgear monitoring tool, nothing crazy.

matmen, 2022-02-22:
I assume it is detecting `probe_speed` as a forbidden string, as it may also be used in malware (network scanning). It's a false positive on Netgear's side.
It's probably fine as-is (I'm not sure if Netgear's software lets you just ignore the alert?), but if it's bothering you you might want to consider reporting it as a false positive to Netgear (or the [corresponding anti-malware service](https://kb.netgear.com/000060943/How-do-I-report-false-positives-detected-by-antivirus-on-my-ReadyNAS-OS-6-storage-system)).

closed 2022-02-22 by matmen