fluidd-core/FluiddPI #22: [Question] Data storage location for different hosts

AndreasReitberger, 2022-01-15:
I'm trying to build a client application to work with Fluidd.
I'm mostly done with the project, however I cannot figure out where FluiddPi stores the different hosts information?

[image]

MainsailOS is doing this in the database, however this seems not to be the case for FluiddPi.
Can you give me a hint? I need to get the information via WebSocket or REST-API.

Thank you!

matmen, 2022-02-22:
Hey, what data are you trying to access specifically?

matmen, 2022-02-22:
The host addresses aren't available through an API (yet); they are stored in the browser's local storage (`localStorage.appInstances`)

matmen, 2022-03-28:
I'll go ahead and close this. Feel free to re-open if you need any additional information 🙂

closed 2022-03-28 by matmen