fluidd-core/FluiddPI #27: Multiple printer setup only shows camera of directly connected IP

ercdvs, 2022-03-08:
I have 2 printers running each with a full fluiddpi/mainsail / klipper install on a RPi Zero 2 on each.
ip address 10.1.1.81 and 10.1.1.102.

Each IP has the other printer installed under 'Printer' on the right hand menu
Each printer also has a USB webcam installed and activated via fluiddpi settings and edited webcam.txt config.

When I connect to 1 IP (.81) I can see the camera for the .81 printer.  when I switch using the right hand menu to the .102 printer, I have the correct printer menu and UI, but the camera shown is for the .81 printer (but not the inverted settings, etc)

This is also true when i directly connect to the .102 printer.  I can see the .102 camera and only that camera.   If one camera is disabled, then I am unable to view the working camera from the other ip address.

Cliff Notes :  Webcam only works on the directly connected IP address for the printer in fluiddpi

matmen, 2022-03-11:
Hey, can you attach a screenshot of your camera settings in Fluidd?

ercdvs, 2022-03-11:
from .81 printer:
[image]

the fluiid webpage when directly connected to .81
[image]

connecting using the right side pane to my .102 printer, all information is correct except the camera view 
[image]

switching directly to .102 i see the same good printer info, but now the correct camera 
[image]

closed 2022-03-18 by matmen