fluidd-core/FluiddPI #28: Fluiddpi missing input_raspicam.so

ilektron, 2022-03-11:
Something is happening causing mjpg-streamer to be compiled without the input_raspicam plugin due to unmet dependencies. If I had to guess, something changed with the location of /opt/vc/include that isn't being installed on the root image causing the plugin to have unmet dependencies. The raspi camera doesn't work out of the box because it's missing the so associated with the plugin.

thisiscam, 2022-03-21:
FWIW, I tracked down to this line: https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_raspicam/CMakeLists.txt#L2, and understood what @ilektron was saying.

ilektron, 2022-03-21:
As a workaround I was able to just change the CMakeLists.txt for the input_raspicam to look in /usr/include/interface for the required library headers and then update all the /opt/vc filepaths to /usr/include/interface. After the change and recompiling, make install, everything things to be working.

I think in the latest releases of pi base images the pi specific headers were moved from /opt/vc into /usr/include

alucryd, 2022-06-09:
Gave the libcamera fork a try, afraid it's not a viable option at the moment, even at 640x480 it completely pegs a 3B+, let alone a Zero 2. UVC uses so little CPU, even at 1080p on a NoIR v2 that I don't think raspicam will be missed.

kbingham, 2022-06-10:
@alucryd Did you try https://github.com/ArduCAM/mjpg-streamer ? or something else?

alucryd, 2022-06-10:
That's the one I tried, yes.

kbingham, 2022-06-10:
Could you tell me what you mean by 'pegs a 3b+'?