fluidd-core/FluiddPI #29: Display error

AntoszHUN, 2022-03-27:
The error persists on the display. I don't know where to write. The display will be streaked if I wake up. Sometimes the image flickers as you live. Resolution cannot be selected after using the raspi-config command. The resolutions do not include the Raspberry Pi 7" Touch Display TFT - DSI connect. Resolution which would be 800x480. FluiddPi is installed. There was nothing wrong with Mainsail, it worked flawlessly. Who should I write to?
[image]

matmen, 2022-03-28:
Unfortunately, I don't have a display for my Pi to troubleshoot this, so someone with a similar setup will have to chime in.

AntoszHUN, 2022-03-28:
Oh, I see. Maybe someone else has this kind of error.