fluidd-core/FluiddPI #30: Asks for login after initial install

ChSt15, 2022-04-03:
Im trying out fluiddpi for the first time so this might just be a beginner issue.
When connecting for the first time it asks for username and password, which I haven't setup yet. Using SSH I changed the moonraker.conf force_logins to false and even after rebooting still asks for login. Using an older version like 1.16.2 solves the issue even after updating everything.

matmen, 2022-04-06:
I've diagnosed this to be an issue with IPv6 (introduced in #9). The moonraker trusted hosts currently only include link-local addresses, but SLAAC (/ public) v6 addresses are required here too. I'm not sure how to fix this ATM, considering that we neither want to just add all hosts to the trusted hosts, nor can we make NGINX listen on link-local only (and I also don't want to just disable IPv6...). Maybe someone else has an idea here? @fluidd-core/triage

heregoesnothing3d, 2022-04-13:
yes, same problem

matmen, 2022-04-13:
A temporary workaround is to use the `1.16.2` FluiddPi image or removing the `listen [::]:80` from the nginx config. We're working on a solution.