fluidd-core/FluiddPI #33: Read-Only filesystem?

ganzgustav22, 2022-04-20:
Would you be interested in making the FluiddPi image a readonly filesystem? Would make things less error-prone on powerloss or people pulling the plug without shutting down and also not wear down the sdcard.

matmen, 2022-04-20:
Hey, I like the idea of having the SD card more protected, but how would this work exactly? The log files, gcode files and moonraker database would still need to be written to the card and be persistent, so the same chance of corruption exists there, no? A PR to extend the lifetime of SD cards by writing the log files to a RAM FS is already there: https://github.com/fluidd-core/FluiddPI/pull/12

ganzgustav22, 2022-04-20:
There are two issues that are solved with a read-only system:
- Data corruption on powerloss
- sdcard wear due to more or less constant writing of lots of data

Yes, ideally everything should be on a ramfs. But having the logfiles on a ramfs instead of the sdcard is definitely a good start as it already reduces the amount of data written to sdcard by a large amount.

Different approaches are possible.

One could move everything to ramfs and only switch the sdcard to read-write and write the data to it when the user shuts down or restarts the system. Ofcourse then, on powerloss, everything since the last restart or shutdown would be gone. But the sdcard would be safe as it's read-only all the time.

Another option would be to sync the data in the ramfs to the sdcard in regular intervals (and only switch the sdcard to read-write during the syncing) but that would ofcourse bear the risk of corruption again if the user manages to kill the power in the wrong moment (i.e. during sync).

Yet another option would be to keep the sdcard read-only and only switch it to read-write when something actually has to be written. But that would only work for stuff like received gcode files or updates to the printer.cfg and the like (i.e. things that do not get changed/written too often), not for stuff like logs or the moonraker db which is more or less steadily written to.

I haven't looked at the details yet, but I think something like this could make sense:
- Make / and /boot read-only by default
- Write logfiles to ramfs, don't do any syncing on them, if a powerloss occurs they're gone and that's it
- Write gcode files as they come in, maybe even to a different partition (so that / and /boot can be kept read-only)
- Write config files as they are changed
- Moonraker db I'm not sure about, does it also contain data that needs to be persistent?

I can give some help with making the base system read-only as I have already done that with Raspberry Pi OS.