fluidd-core/FluiddPI #34: USB disconnect/timeout issues with Ender 7 with ch341-uart USB2Serial

ganzgustav22, 2022-04-20:


matmen, 2022-04-20:
I'm not sure if this would be a FluiddPi issue. I personally can't reproduce this, though maybe you are willing to troubleshoot: Is this reproducible on a clean Raspberry Pi OS install with Klipper / fluidd installed via KIAUH?

ganzgustav22, 2022-04-22:
It's not a bug in the Fluiddpi components or changes to the base Raspberry Pi OS image, indeed. It's a kernel and/or Pi firmware issue within Raspberry Pi OS. But it affects FluiddPi users, so I thought I report it here so that people can find it when they face the same issue.

I can open an issue on the Raspberry Pi github, but my experience is that these issues often do not get fixed because they are sporadic, hard to debug and need specific hardware. 

Sometimes they get fixed "by accident", when a larger kernel release bump occurs, sometimes they get re-introduced later, etc. it's quite annoying.

What still left to try is using the Pi4 USB-C power connector as a workaround (It's basically the same old USB controller from the older Pis, so there is hope that it works with that one). Need to order a cable first and power the Pi4 via it's GPIO pins. I can report back if that turns out successful.