fluidd-core/FluiddPI #41: [1.18.2] shadow password for user pi invalid

bassco, 2022-06-18:
I downloaded and installed 1.18.2 to see what was wrong with the ssh login.

Mounted the rootfs filesystem to look around and found the issue with the `pi` user that does not have a password set in `/etc/shadow` which is failling all logins.

HTH to find a fix.

## v.1.18.2

```shell
# cat fluiddpi_version
0.4.0

# grep ^pi *shadow? passwd?
gshadow-:pi:!::
shadow-:pi:*:19086:0:99999:7:::
passwd-:pi:x:1000:1000:,,,:/home/pi:/bin/bash
```

😢 - fluiddpi_version is not updated with the tagged version number

## Working version

```shell
cat fluiddpi_version
0.4.0
grep ^pi *shadow? passwd?
gshadow-:pi:!::
shadow-:pi:<REDACTED>:18754:0:99999:7:::
passwd-:pi:x:1000:1000:,,,:/home/pi:/bin/bash
```

matmen, 2022-06-18:
This is a known issue in the 1.18.2 image (that's why it's marked a pre-release; 1.17.0 is fine, though) and has been fixed in the CustomPiOS already. We will release a new FluiddPi version with the next release of Fluidd.
Thanks for reporting the `fluiddpi_version` issue, I'll look into having it updated automatically :)

bassco, 2022-06-18:
Thanks for the feedback. Looking forward to the new release.

FYI: The release notes for 1.18.2 are also pointing to release 1.17.2 which is a broken release link.

I managed to add a password to the shadow file and am testing the rest of the functionality on the image.

closed 2022-06-18 by bassco