fluidd-core/FluiddPI #56: Pause macro causes printer to lose homing

SaschaUncia, 2022-09-30:
Adding the default PAUSE macro to my gcode causes the printer to pause where expected, and if left alone for a couple of minutes, to cool the hotend and bed (this is fine). However it also loses it's homing location, and it won't let me resume the print without rehoming all axis. 

How do I prevent this behavior?

arabisaldrin, 2022-09-30:
idle_timeout. https://www.klipper3d.org/Config_Reference.html#idle_timeout
defaults to turn off heaters and disable motors.

SaschaUncia, 2022-09-30:
Ok, I think I see, You can enter gcode in the [Idle_Timeout] section which should cause the printer to ignore the default behavior. 

Problem is it keeps giving me syntax errors

`[idle_timeout]
#gcode:
#   A list of G-Code commands to execute on an idle timeout. See
#   docs/Command_Templates.md for G-Code format. The default is to run
#   "TURN_OFF_HEATERS" and "M84".
gcode:
M104 S0
timeout: 720
#   Idle time (in seconds) to wait before running the above G-Code
#   commands. The default is 600 seconds.`

Results in a warning of 

> File contains parsing errors: /home/Sascha/klipper_config/printer.cfg
> [line 108]: 'M104 S0\n'
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Printer is halted

Even using the default settings of "TURN_OFF_HEATERS" and "M84" as defined in the comments results in the same error (with or without the quotes).

SaschaUncia, 2022-09-30:
*Sigh*
It needed a space before M104
Syntax errors are the worst...

I'll go try it out